OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #7: Set up new Co2 sensors

SamanthaCra, 2024-07-25:
There are a range of new Co2 sensors that need to be breadboarded and tested.

SamanthaCra, 2024-08-12:
Martin requested as many as possible.

SamanthaCra, 2024-08-28:
[image]
https://github.com/OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa/tree/sc-co2-test

kierenblack, 2024-08-29:
In the TinyLora.h library file, please ensure you add `#define AU915` on line 73 before
```
/** Region configuration*/
#if !defined(EU863) && !defined(AU915) && !defined(AS920)
#define US902 ///< Used in USA, Canada and South America
#endif
```

fuzzyers, 2024-09-01:
[image]

So we have 3 complete sensors as well as 2 breadboarded.
We are waiting on more female header pins to continue with this process.

fuzzyers, 2024-09-02:
We have finished development on sensors for this sprint we await more parts in the future our next steps will include getting the developed ones in cases and then moving them on to the c02 team

closed 2024-09-02 by fuzzyers