OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #20: Get data sending to database

SamanthaCra, 2024-08-05:
Ge tthe sensor data from IoT sending to the database successfully.

SamanthaCra, 2024-08-05:
Similar to Transmit Data LORA #13

SamanthaCra, 2024-08-05:
https://www.thethingsindustries.com/docs/api/

SamanthaCra, 2024-08-05:
It is possible to get HTTP sending but if we are already developing an API then this is the best solution. We will need to build a web hook in order for the api to take data and post it to the database.

fuzzyers, 2024-08-21:
[Backend Wiki](https://github.com/OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa/wiki/Backend)

I have assembled a basic backend capable of ingesting data for temperature and humidity as a prototype.

Aswell as a very simplistic Webhook although it's only been setup to a local PostgreSQL database.

closed 2024-08-28 by fuzzyers