OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #27: Come up with a solution as to how we can house the weather station on the roof

SamanthaCra, 2024-08-08:
Now that we have access, we need to put a plan together of requirements for our sensors and how they will come together on the roof.

kierenblack, 2024-08-12:
The pole can be mounted using a pipe clamp

kierenblack, 2024-09-02:
Rob is looking into some pipe and scaffold clamps to clamp the weather station to.

fuzzyers, 2024-09-07:
i have found this [waterproofbox](https://www.mitre10.co.nz/shop/cordtech-weatherproof-power-box-345-x-210-x-125mm/p/270766?store=28&gad_source=1&gclid=EAIaIQobChMIpoL4gIKyiAMVkSiDAx0rBzeTEAQYAiABEgLtNvD_BwE&gclsrc=aw.ds) with cable access sold at mitre 10

kierenblack, 2024-09-15:
I have left a length of pipe with 2 clamps in the hardware room to be used when the weather station is ready to go up to the roof.

fuzzyers, 2024-11-01:
[image]
We have settled on one of these waterproof boxes from mitre 10 which we have spray painted white to help keep them cool

closed 2024-11-01 by fuzzyers