OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #109: Create ability to switch between week view and day view on graphs

BenCampbellNelson, 2024-10-22:
It would be great to able to view data for the day on one view type then expand that out to show the whole week.

SamanthaCra, 2024-11-08:
Function created: https://github.com/OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa/wiki/Database-Server

```
CREATE OR REPLACE FUNCTION get_daily_avg_data(table_name TEXT, column_name TEXT, days_interval INT)
RETURNS TABLE(day DATE, day_of_week TEXT, avg_value DECIMAL(7,2)) AS $$
DECLARE
    query TEXT;
BEGIN
    -- Check if days_interval is either 7 or 30; if not, raise an exception
    IF days_interval != 7 AND days_interval != 30 THEN
        RAISE EXCEPTION 'Invalid days_interval: %; only 7 or 30 are allowed.', days_interval;
    END IF;
    query := format(
        'SELECT 
            DATE_TRUNC(''day'', timestamp)::DATE AS day,
            TO_CHAR(timestamp, ''Day'') AS day_of_week,
            ROUND(AVG(%I), 2)::DECIMAL(7,2) AS avg_value
         FROM %I
         WHERE timestamp >= NOW() - INTERVAL ''1 day'' * ($1 - 1)
         GROUP BY day, day_of_week
         ORDER BY day DESC',
        column_name, table_name
    );

    -- Execute the dynamic query and return the result
    RETURN QUERY EXECUTE query USING days_interval;
END;
$$ LANGUAGE plpgsql;
```

SamanthaCra, 2024-11-08:
[image]
[image]
[image]

closed 2024-11-08 by SamanthaCra