OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #172: Set up local postgres database with fake data

TylorJB, 2025-03-31:
In order to run the backend/database locally we need data. So I have spoken with my team and we are going to set up a postgres database with fake data. What I plan on doing: 

1. Reading documentation on the database schema and functions
2. Create Fake data 
3. Run locally 
4. Share with my team 
5. Write documentation

TylorJB, 2025-05-01:
I have successfully created fake data as is shown below.

[image]

I wrote a text file explaining the steps I took to complete it. 

[image]

My teammate Bradley tested it and it worked. 

I then added to the wiki

[image]

closed 2025-05-01 by TylorJB

Robebj2, 2025-05-01:
This works well. I can now run tests locally using the fake data.