OtagoPolytechnic/Cloudy-with-a-Chance-of-LoRa #217: final design of the station box

Sophie-skyyye, 2025-05-30:
The purporse of this issue is to track th eprogress of the final build.

we had a design in mind
- we wanted the box to be made of wood
- have vent holes 
- have a slanted roof 
- have a hinged door with a latch 
- better cable management 
- water proof

Sophie-skyyye, 2025-05-30:
Drew the measurements onto the wood and stated cutting 

[image]

Sophie-skyyye, 2025-05-30:
Once everything was cut out i cut wholes where the vents would go and then kayla sanded things down and statred painting with the special water proof paint we got.

[image]

[image]

Sophie-skyyye, 2025-05-30:
The Vents are now dry and secured to the front and sides. We then put the walls together.

[image]

[image]

Sophie-skyyye, 2025-05-30:
The roof was painted a lot as it was going to be in the rain the most so once it was dry we attempted to put it on however, I think i was so gassed out by the fumes of the paint and silicon that i made the roof to small. The roof then got converted in to the floor because it fit so well. 
so a new roof got made. 

[image]

[image]

Sophie-skyyye, 2025-05-30:
we had to fill a lot of gaps with silicon as my wood cuts were a lil wonky

kaylapatt71, 2025-06-03:
[image]

Door and latch has been added

kaylapatt71, 2025-06-03:
[image]

Hole for cords has been added to the bottom of the station and some feet for airflow/cord access

Sophie-skyyye, 2025-06-09:
The box is on the roof and completed, the box was also up on the roof for the weekend while it was raining and after inspecting the inside it was completely dry

[image]

[image]

[image]